Xinyuan-LilyGO/T-Display-S3-Long #1: So many issues...

d3mac123, 2023-11-07:
Hi, just got my S3-Long. I followed all the instructions for the Arduino IDE Quick Start. As the instructions are not that clear, 

(step 2) after installing the Arduino ESP32 (is the Arduino_ESP32_OTA the right one?), 
(steps 3) I unzipped the downloaded file (the entire github from https://github.com/Xinyuan-LilyGO/T-Display-S3-Long/tree/master), and I copied it to my Documents\Arduino\libraries folder
(step 4) I copied the lib folder to my Documents\Arduino\libraries folder as well - shouldn't this be copied to \Library\Arduino15\libraries (just in case, I did it too)
(step 5) Lastly, I configured the Tools as specified.

Issues:
(step 6) The T-Display-S3-Long does NOT show in File->Examples->...
(step 7) The board does not show in some of my USB ports. However, I made it to work in one port at least
(step 8) I manually copied the touch example and uploaded it to the board - the display does not show anything (all black) but I could capture the touch via serial monitor;

No other example worked - I tried Step 9 several times.

Questions are:
- What am I missing?
- Can the board run code using the TFT_eSPI library?

ShallowGreen123, 2023-11-08:
> Hi, just got my S3-Long. I followed all the instructions for the Arduino IDE Quick Start. As the instructions are not that clear,
>
> (step 2) after installing the Arduino ESP32 (is the [...]

Sorry for the error in the previous Readme, now I have changed the new readme;
The right thing to do is：
1. Download the “S3-Long” project
2. copy the entire “ S3-Long/lib” folder to the arduion Lib (e.g. C:\Users\YourName\Documents\Arduino\libraries)
3. open any one with Arduion IDE “S3-Long/examples/XXX XXX. Ino”
4. Select ESP32S3 Dev Module and port
5. Click "upload" button to update the code to the board

nikthefix, 2023-11-17:
Hi d3mac123,

If you're still having problems then I suggest backtracking a bit and starting with the TFT demo in the Examples folder. Open it directly from the folder. Since this demo does not require anything to be copied to your Arduino Lib folder or use any screen rotation (it's in portrait mode) it should work out-of-the-box as it did for me. At least this will allow you to test the display hardware. You already know that touch is working from your previous experiments.

The other examples took a bit more work for me as I needed to use the latest LVGL version (not the supplied LVGL configuration) and the latest 5.xx IDF Arduino ESP32 iteration. Both have been recently updated and in some cases there are breaking changes which affect example code supplied with new boards. The dust will settle soon I'm sure but it must be difficult for Lilygo and others when the goalposts keep changing. I'll post some code mods here asap. I only got the board yesterday so I'm experimenting too.

d3mac123, 2023-11-17:
Nik, thanks for the help. Yes, I got the examples to work but, as of now, the board is in "waiting mode" :) I have wrote more than 1000 lines of code using the TFT_eSPI library and I cannot transfer the work to the Long version. I have tried the library you made for Volos' AMOLED S3 but it did not work so, until you do your magic, I am not using the Long version.

Thanks again!

nikthefix, 2023-11-17:
Hi again. Yes the AMOLED code won't work with the Long as it uses a TFT driver even tho both are set up for QSPI by default. So am I right in thinking that you would want to use the Long in landscape mode with TFT_eSPI?
 I also received their 2.41" AMOLED T4 this morning. Very nice hardware. It has similar software versioning issues (at least on my computers).

d3mac123, 2023-11-17:
You are right (my project requires landscape mode with the TFT_eSPI).

I have to check yet the T4.

Alex Souza

nikthefix, 2023-11-29:
Hi d3mac123,

I will upload some new demo code here in a couple of days. Could you try it and give me some feedback if you get a moment?
It's only for the Long / TFT_eSPI V2.5.34 / Arduino ESP32 V3.0.0-alpha2 as I haven't tested it with other versions.
It's stripped right back so is not a drop-in replacement for the supplied libs.
I'm just tidying it up a bit right now.

nik

d3mac123, 2023-11-29:
Sure thing. Looking forward to that!

Alex Souza

nikthefix, 2023-11-30:
Hi d3mac123,

Not a demo but a hello world. Maybe a good starting point. I'll pull something a bit more useful asap. No touch in this. But you've got that covered.

[TFT_Long_eSPI_hello_world.zip](https://github.com/Xinyuan-LilyGO/T-Display-S3-Long/files/13515021/TFT_Long_eSPI_hello_world.zip)

nikthefix, 2023-11-30:
Much of what remains in my driver mod is redundant if you're using TFT_eSPI. I just need to trim it down a bit more.

d3mac123, 2023-11-30:
I tried the code with my current setup (Esp32 2.0.14 and TFT_eSPI 2.5.34), and it worked fine (thank you!).

As I am working with the board with the USB port on the left side (using the lcd_setRotation(2);), I will work on the touch part now (as touching the bottom right of the screen is giving me coordinates x=0,y=180, instead of the right ones, which should be x=640,y=180).

Please keep me posted on your improvements. Your work is really appreciated!

nikthefix, 2023-11-30:
The greatest optimization in terms of screen refresh will come from identifying which elements of the sprite are updated and only pushing those regions to lcd_PushColors_rotated_90.  At present the whole display is refreshed which is grossly inefficient.

This will make all the difference considering that we're already compromised having to use a soft rotate. 

There is also a hardware accelerated colour fill (which I will implement) that allows a mega fast clear screen along with various test patterns. The datasheet is actually very interesting.

All the best.

nik

d3mac123, 2023-12-01:
@nikthefix I have noticed a glitch in the top right side of the display (see pictures). Basically, some lines that are not in the code have been drawn over the green circle. I saw it also with different code (that area is always "glitching some lines").

Something for you to double-check?
[display_glitch.zip](https://github.com/Xinyuan-LilyGO/T-Display-S3-Long/files/13531327/display_glitch.zip)

nikthefix, 2023-12-02:
Try reducing the spi frequency from 32000000 to maybe 20000000 and see if that helps. If not then it sounds like the rotate buffer is overflowing. Increase 

uint16_t* qBuffer = (uint16_t*) heap_caps_malloc(203400, MALLOC_CAP_SPIRAM ); //psram buffer for matrix rotation (640 * 180 * 2)

from 203400 to 203420 as this is the exact size of the TFT_eSPI sprite buffer.

I will try to reproduce the glitch.

nikthefix, 2023-12-02:
Yes forget the spi thing - it's the gBuffer being undersized and it needs more than 20 extra bytes. Try 
220000, MALLOC_CAP_SPIRAM for the time being and i'll work out the correct size and report back.

nikthefix, 2023-12-02:
Hi again. It turns out it was a typo. 203400 should have been 230400 (600 x 180 x 2). My bad.

d3mac123, 2023-12-02:
It works fine! Thanks!

d3mac123, 2023-12-07:
@nikthefix (please let me know if you prefer to start a new thread elsewhere). I have noticed that when using the WifiManager library, my board keeps restarting the setup function. Everything works fine in the regular T-Display board but not with the Long. Any ideas?

nikthefix, 2023-12-08:
@d3mac123 Someone else mentioned this in regards to a different board (RGB parallel display) and I also noticed it with the new T4-Amoled. I'll look into it.

nikthefix, 2023-12-19:
@d3mac123 

Hello again. 

How's it going with the Long?
I've been working with Volos on a new touch demo for the Long and I think he will publish a video and code in the next day or so.
Watch his space. I think it will be good.

I've also added fast partial updates for LVGL projects - compatible with alpha 3 and below.
I'm just integrating those changes to work with TFT_eSPI so we can have fast animations.
I will post on my github and here.

All the best,

nik

momosh13, 2023-12-19:
Hi. Please tell me how to setup lvgl? I'm having lot's of problems compiling the factory.ino file for T-Display-S3-Long.
Please help

nikthefix, 2023-12-19:
Hi momosh13,

Try the attached sketch and see if it works. The buttons do nothing at the moment.

Use a fresh LVGL with mods to lv_config as described in the sketch.
If it works OK then maybe transfer the mods and libs to the lilygo factory example as substitutes.

[tft_long_lvgl_odometer.zip](https://github.com/Xinyuan-LilyGO/T-Display-S3-Long/files/13714328/tft_long_lvgl_odometer.zip)

momosh13, 2023-12-19:
Thank you for the reply. the lvgl is 8.3.0 yes?

nikthefix, 2023-12-19:
8.3.11 for my tests.

d3mac123, 2023-12-19:
Thanks fon the updates @nikthefix . Project is on hold waiting for the touch example but also due to the network issues I mentioned in another post (board keeps resetting while using the Wi-Fi Manager library).

I'm looking forward for Volo's example!

nikthefix, 2023-12-19:
Yeah that's definitely not your fault. The wifi manager is not working for me either. Trying to find a work around. 

Further updates may be found here:

https://github.com/nikthefix?tab=repositories

But I'll stick them here too for the time being.

Is everything else working? GUI, touch etc?
The touch 'double tap' problem you reported should be solved in the upcoming Volos code. Let me know. It should be as we worked hard to get touch working in alpha1/2/3

The pictures you sent of your interface looked really good.

nik

nikthefix, 2023-12-19:
Looks great! 
Watch this space.

nik

momosh13, 2023-12-20:
> 8.3.11 for my tests. But it's important NOT to use the lilygo supplied libs as they have been modified and will not work with my code. It's not their fault. The Arduino_ESP32 core updated recently and there were many breaking changes.
> 
> If you're using alpha3 then you'll need a workaround for PWM to control back-light brightness as this is not working at present but you can do it manually with a timer.
> 
> In my demo the backlight is just set to fully on.

Thank you for your help, finally your sketch working good on T-Display-S3-Long, but the factory sketch not working, I have this error : "text section exceeds available space in board"  when Partition Scheme on Default 4MB. When I'm changing the Partition Scheme to 16M Flash / FATFS, NO more error but the screen stays black, no image, Please help me to fix this.

momosh13, 2023-12-20:
The screen showing this

[image]

momosh13, 2023-12-20:
yours, thank you 

[image]

momosh13, 2023-12-20:
This is my settings

[image]

nikthefix, 2023-12-20:
@momosh13,

The factory demo needs the supplied lilygo modified libs. Unfortunately these mods don't seem to be working with any of the esp-arduino cores that I've tried (2.5 to alpha3).

I've started reworking the demo to use alpha3 and standard lvgl. You can try what I've done so far. The wifi stuff is commented out so the clock won't flip yet as there's no ntp time reference. 

#define LV_USE_MSG 1 in lv_conf.h line 706 in addition to the other lv_config changes you made before.

[flip_clock_long.zip](https://github.com/Xinyuan-LilyGO/T-Display-S3-Long/files/13723673/flip_clock_long.zip)

momosh13, 2023-12-20:
Thank you. but I still getting this:

Sketch uses 1312829 bytes (100%) of program storage space. Maximum is 1310720 bytes.
Global variables use 111068 bytes (33%) of dynamic memory, leaving 216612 bytes for local variables. Maximum is 327680 bytes.
Sketch too big; see https://support.arduino.cc/hc/en-us/articles/360013825179 for tips on reducing it.
text section exceeds available space in board

Compilation error: text section exceeds available space in board

nikthefix, 2023-12-20:
Check flash size and partition setting in build options. Should be:
16MB and 16M flash (3M APP/9.9MBB FAT FS)

You should see:

Sketch uses 1482237 bytes (47%) of program storage space. Maximum is 3145728 bytes.
Global variables use 58096 bytes (17%) of dynamic memory, leaving 269584 bytes for local variables. Maximum is 327680 bytes.

momosh13, 2023-12-20:
yes, you're right, now is (41%) of program storage space, but now I'm back to the original problem, as soon as I'm choosing Partition Scheme 16M flash (3M APP/9.9MBB FAT FS) the screen gouse black, no picture

nikthefix, 2023-12-20:
Check psram is set to OPI.
Check lv_conf.h has all settings described in the comments of the odometer sketch.
Check that XPowersLib is in your libraries folder. Install latest version if it's old.
Check that you have ESP32 3.0.0 alpha3 installed.

nikthefix, 2023-12-20:
@d3mac123 
@momosh13 

I'll put future updates at https://github.com/nikthefix

Raise further issues there if that's OK with you.

Volos will publish his new video today at 12 noon UK time. I will make those files public at that time. I didn't want to release anything before his video went out.

momosh13, 2023-12-21:
Please, fix the factory file, I really like that clock, Volos says you are a genius, only you can do it.

nikthefix, 2023-12-21:
@momosh13 I really like that clock too! I will work on it today.

momosh13, 2023-12-21:
no, no lock with  ESP32 3.0.0

nikthefix, 2023-12-21:
OK well we just need to compare your setup with mine.
Could you get the odometer demo running again just to double check that nothing else has changed.

For the flip clock:

ESP32 3.0.0 alpha3 via board manager. (https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_dev_index.json)

LVGL 8.3.11 via library manager.

XpowersLib 0.2.1 via library manager.

Remove anything you might have installed from the original supplied lilygo support files.
Reconfigure the build options in Arduino IDE as the alpha3 update will reset them:

  Board  ESP32-S3-Dev
  USB CDC On boot Enabled
  Flash Size 16MB
  Partition Scheme 16M Flash(3MB APP/9.9MB FATFS)
  PSRAM "OPI PSRAM"

  Check lv_conf.h options:

  Line 15 - #if 1 (Set it to "1" to enable content)
  Line 30 - #define LV_COLOR_16_SWAP 1
  Line 49 - #define LV_MEM_CUSTOM 1
  Line 88 - #define LV_TICK_CUSTOM 1
  Lines 354 to 384 - enable fonts
  Line 706 - #define LV_USE_MSG 1 

I have the clock factory demo working now with the above setup (zip attached).
Set your SSID and PASSWORD in the sketch at lines 22 and 23.
Set your timezone offset in pins.h

I've not tested the power manager yet and the wifi-auto-config is disabled at the moment. But the clock flips and tells time - at least it does here! :)

Let me know if it works.

[flip_clock_long.zip](https://github.com/Xinyuan-LilyGO/T-Display-S3-Long/files/13740241/flip_clock_long.zip)

momosh13, 2023-12-22:
Thank you for your help. Only odometer works perfectly here, with 16M Flash(3MB APP/9.9MB FATFS) or 4MB, the flip_clock_long and the factory not working, when Partition Scheme is 4MB says file is too large, when Partition Scheme is 16M or 32M screen goes Black.
Maybe my computer(MAC) is the problem?, I'm going to transfer everything to my windows computer and let you know 
Sorry

https://github.com/Xinyuan-LilyGO/T-Display-S3-Long/assets/5689051/17f9e433-0ad8-4218-9724-dc367cb9af3d

nikthefix, 2023-12-22:
Also, see if there's anything here which might help.

https://github.com/nikthefix/Lilygo_Support_T_Display_S3_Long_TFT_eSPI_Volos-nikthefix/issues/1

momosh13, 2024-01-06:
Hi
First, Happy New Year !!!
Please, see the attached video, I able to get here by making this two changes, to Oreginal code,
1. #define SPI_FREQUENCY           32000000  to  #define SPI_FREQUENCY         10000000
2.  commenting out this  //disp_drv.full_refresh = 1;          // full_refresh must be 1
it doesn't showing the numbers and little destortions around the push buttons,
Please, let me know, if you had a progress with this clock.
Thank you  

https://github.com/Xinyuan-LilyGO/T-Display-S3-Long/assets/5689051/22c2f2b3-a3bd-42a8-a2f8-e5d1e8f624ee

nikthefix, 2024-01-06:
Happy New Year to you too!

Glad you got it working.

Yes the 32MHz is too fast for this display when you are not using disp_drv.full_refresh.
10MHz is what I use.

As for the distortion, this is caused by the display only accepting 'even' partial update regions. You will see corruption if you update a sub-region of odd numbered width or height. If you look at my odometer demo and find the callback 'my_rounder', this is a routine which quantizes regions to the nearest even numbers and should solve the problem in LVGL with the Long.
This is only an issue if full refresh is off - but full refresh is slow.

Just import the callback to your original factory demo and add:

disp_drv.rounder_cb   = my_rounder;      to the driver setup parameters.

My updated version of the flip clock demo has this implemented already.

nikthefix, 2024-01-06:
BTW, my updated version of the flip clock demo is at my github:
[nikthefix](https://github.com/nikthefix?tab=repositories)

momosh13, 2024-01-15:
Thank you nikthefix !!!
You're genius.
The flip_clock working on my display, perfectly.

nikthefix, 2024-01-15:
Nice one!
Thanks for letting me know.

nik

momosh13, 2024-01-18:
Hi.
I have one more question about the clock, now, it's a 24Hours clock, there is any way you make 12House clock option?
Thank you 

https://github.com/Xinyuan-LilyGO/T-Display-S3-Long/assets/5689051/5d65c2f5-9a05-4c57-8fab-9a97e42b5f4b

nikthefix, 2024-01-19:
Yes it should be possible to adjust the format. I'll try it tomorrow and get back to you here.

nik

nikthefix, 2024-01-22:
@momosh13 

Hi again. Regarding 12hr format for the flip clock...
The simplest way is to retain internal 24hr time keeping and change what's displayed:

In factory_gui.cpp,
In function void set_flip_time_anim(int hour, int minute, int second) , 
At line 74 add:

    if (hour>12) hour=hour-12;

At least this way you don't have to modify any libs and your code will work even if you change your time reference (such as, maybe in future you'd prefer to use a RTC or GPS).

nik

momosh13, 2024-01-23:
You did it  !!!
It's 12H Clock now.
Thank you

nikthefix, 2024-01-23:
@momosh13 Good . I will add it to my github. Thanks for pointing it out.
nik

momosh13, 2024-03-25:
Hi nikthefix
If you're still in this forum, I have a question for you, if I may?
I don't know anybody, who my help me, but you:)
I'm working on this project many months and almost no progress, I'm trying to send HEX command from Arduino / USB Host to USB device to control some functions. I tried it, Teensy 4.1 with USBHost_t36 library, and Arduino USB Host Shield 2.0 with https://github.com/felis/USB_Host_Shield_2.0 library, no luck.
I need to send this HEX

1b 00 a0 24 08 16 0a d7 ff ff 00 00 00 00 09 00 00 03 00 17 00 02 03 53 00 00 00 00 40 00 40 48 48 40 00 00 40 00 00 40 00 00 40 00 00 40 00 00 40 00 00 40 00 00 40 00 00 40 00 00 40 00 08 48 08 08 48 08 00 40 00 08 48 08 00 40 00 08 48 08 00 40 00 00 40 00 00 40 00 00 40 00 00 40 00 00 40 00 08 48 08 00 40 00 00 40 00 00 40 48

to (Vendor Specific) USB Device. I got the HEX, by sniffing communication between Windows control software and the device, with Wireshark.
Please, maybe you can help me again.
Thank you

nikthefix, 2024-03-25:
@momosh13 
I have some of these USB host boards so I can look into it. Which board are you using?
There is the Uno shield and the smaller design from here:

https://circuitsathome.com/usb-host-shield-for-arduino-pro-mini/

There are clones of the 'mini' version which are slightly different in the power delivery department.

What does the vendor specific device enumerate as in your computer device manager?
Does it use a custom USB driver or is it class compliant?

I have the host shield working fine with the CDC USB of my laser cutter GRBL controller, and I've had it working well with keyboard  / mouse.

The ESP32-S3 can also act as host natively.

https://github.com/tanakamasayuki/EspUsbHost

nik

nik

momosh13, 2024-03-26:
Thank you for reply 
I have two hardwere otions and i have no problem connecting to my device and read the info from it, the device is USB-HID.
1. i have teensy 4.1 board, please see the picture
[image]

and with  https://github.com/PaulStoffregen/USBHost_t36 library, example [HIDDeviceInfo](https://github.com/PaulStoffregen/USBHost_t36/tree/master/examples/HIDDeviceInfo)
the serial monitor giving me this
[image]

2. then i have Arduino USB Host Shield 2.0 with theensy 4.0, please see the picture
[image]

whit this one,  https://github.com/felis/USB_Host_Shield_2.0 library, exsample [USB_desc](https://github.com/felis/USB_Host_Shield_2.0/tree/master/examples/USB_desc), i'm geting this

[image]

So, my problem is, i don't now how to send the HEX command to my device, i have no idea.

About the  HEX in the previous post. 
in the  Wireshark, i see this 

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

momosh13, 2024-03-26:
I just got one of this to, if you thing, this will make it easier 

[image]

nikthefix, 2024-03-26:
@momosh13 

OK all that info really helps.

First I'd try the Teensy setup and run the example: 

https://github.com/PaulStoffregen/USBHost_t36/tree/master/examples/RawHID/RawHIDtoUSBHostRawHID

Copy the VID and PID from your HIDDeviceInfo dump into the example sketch and start experimenting with Report Descriptors from there.

Here's a useful tool for Windows which will allow you to test data strings to confirm functionality:

 https://www.researchgate.net/figure/HID-terminal-program-used-to-send-commands-to-micro-controller_fig11_268323254

I think it's important to establish working HID comms from the PC using the HID terminal before migrating to the MCU as it'll probably save a lot of time in the end.

nik

nikthefix, 2024-03-26:
@momosh13 

https://github.com/nikthefix/ESP32-S3-USB-HOST/issues/1

SirusPTB, 2025-11-16:
> Hi d3mac123,
> 
> Not a demo but a hello world. Maybe a good starting point. I'll pull something a bit more useful asap. No touch in this. But you've got that covered anyway.
> 
> Please read the comments in the sketch as there's a lot to talk about.
> 
> [TFT_Long_eSPI_hello_world.zip](https://github.com/Xinyuan-LilyGO/T-Display-S3-Long/files/13515021/TFT_Long_eSPI_hello_world.zip)

Thanks for the working example. One question do you know a way to change the fonts?