Xinyuan-LilyGO/T-Display-S3-Long #3: gpio: gpio_set_level(227): GPIO output gpio_num error

d3mac123, 2023-11-07:
Trying to run a simple code (with the settings suggested in this git) and getting the following error:
`E (19) gpio: gpio_set_level(227): GPIO output gpio_num error
 `

ideas? My code is:
```
#include <TFT_eSPI.h>
TFT_eSPI tft = TFT_eSPI();
TFT_eSprite sprite = TFT_eSprite(&tft);
#include "fonts.h"

#define din18 DIN_Condensed_Bold18pt7b

void setup() {
  Serial.begin(115200);
  Serial.println("Starting");

  tft.init();
  tft.setRotation(1);
  sprite.createSprite(320,170);
  sprite.setTextColor(TFT_WHITE, TFT_BLACK);
  sprite.setTextDatum(4);
  sprite.setFreeFont(&din18);
  sprite.fillSprite(TFT_BLACK);

  sprite.drawString("ALTITUDE",320/2,170/2);
  sprite.pushSprite(0,0);
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

ShallowGreen123, 2023-11-08:
The driver of the T-Display-S3-Long screen is AXS15231B and uses the four-wire SPI, which is not supported by TFT_eSPI.

d3mac123, 2023-11-08:
Damn it :( I spent months coding my app with the TFT_eSPI in the hopes the S3-Long would use the same. I guess I will have to keep using the regular T-Display while learning how to (re)code using the new driver...

closed 2023-12-03 by d3mac123