Xinyuan-LilyGO/T-Display-S3-Long #4: I2C wire

d3mac123, 2023-11-10:
Hi, I following the examples and I am trying to add my two I2C sensors to the samples. Usually, I enable them with:
`Wire.begin(43,44)`

However, this is not working with the board. I have tried all the options below but none has worked so far:
`Wire.begin(TOUCH_IICSDA, TOUCH_IICSCL); //enables only the screen touch`
`Wire.begin(43,44); //does nothing`
`Wire.begin(); //does nothing`
`Wire.begin(TOUCH_IICSDA, TOUCH_IICSCL, 43, 44); //does nothing`

Any ideas?

lewisxhe, 2023-11-29:
How are you connected? Wire has been assigned to the touch, normal initialization using external Qwiic is Wire1.begin(SDA PIN,SCL PIN);

d3mac123, 2023-11-29:
I'm using a qwiic sensor connected to the top (43,44) board connector.

lewisxhe, 2023-11-29:
Can you paste your test code? Let me know how you use it

d3mac123, 2023-11-29:
This is the code I am using (connecting to a TMP102 temperature sensor via Qwiic cable) - it is basically yours example with the connection with the sensor via pins 43,44
```
#include <Arduino.h>
#include <Wire.h>
#include "AXS15231B.h"

#include <SparkFunTMP102.h>
TMP102 sensor0;

uint8_t ALS_ADDRESS = 0x3B;
#define TOUCH_IICSCL 10
#define TOUCH_IICSDA 15
#define TOUCH_INT 11
#define TOUCH_RES 16

#define AXS_TOUCH_ONE_POINT_LEN             6
#define AXS_TOUCH_BUF_HEAD_LEN              2

#define AXS_TOUCH_GESTURE_POS               0
#define AXS_TOUCH_POINT_NUM                 1
#define AXS_TOUCH_EVENT_POS                 2
#define AXS_TOUCH_X_H_POS                   2
#define AXS_TOUCH_X_L_POS                   3
#define AXS_TOUCH_ID_POS                    4
#define AXS_TOUCH_Y_H_POS                   4
#define AXS_TOUCH_Y_L_POS                   5
#define AXS_TOUCH_WEIGHT_POS                6
#define AXS_TOUCH_AREA_POS                  7

#define AXS_GET_POINT_NUM(buf) buf[AXS_TOUCH_POINT_NUM]
#define AXS_GET_GESTURE_TYPE(buf)  buf[AXS_TOUCH_GESTURE_POS]
#define AXS_GET_POINT_X(buf,point_index) (((uint16_t)(buf[AXS_TOUCH_ONE_POINT_LEN*point_index+AXS_TOUCH_X_H_POS] & 0x0F) <<8) + (uint16_t)buf[AXS_TOUCH_ONE_POINT_LEN*point_index+AXS_TOUCH_X_L_POS])
#define AXS_GET_POINT_Y(buf,point_index) (((uint16_t)(buf[AXS_TOUCH_ONE_POINT_LEN*point_index+AXS_TOUCH_Y_H_POS] & 0x0F) <<8) + (uint16_t)buf[AXS_TOUCH_ONE_POINT_LEN*point_index+AXS_TOUCH_Y_L_POS])
#define AXS_GET_POINT_EVENT(buf,point_index) (buf[AXS_TOUCH_ONE_POINT_LEN*point_index+AXS_TOUCH_EVENT_POS] >> 6)

void setup() {
    Serial.begin(115200);
    Serial.println("sta\n");

    pinMode(TOUCH_RES, OUTPUT);
    digitalWrite(TOUCH_RES, HIGH);delay(2);
    digitalWrite(TOUCH_RES, LOW);delay(10);
    digitalWrite(TOUCH_RES, HIGH);delay(2);

    Wire.begin(TOUCH_IICSDA, TOUCH_IICSCL);
    Wire1.begin(43,44);

    if(!sensor0.begin())
    {
      Serial.println("Cannot connect to TMP102.");
      //while(1);
    } else { Serial.println("Connected to TMP102.");}

    axs15231_init();  //If you use touch, you must initialize the screen first

    Serial.println("end\n");
}

uint8_t read_touchpad_cmd[11] = {0xb5, 0xab, 0xa5, 0x5a, 0x0, 0x0, 0x0, 0x8};
void loop() {
    uint8_t buff[20] = {0};
    Wire.beginTransmission(ALS_ADDRESS);
    Wire.write(read_touchpad_cmd, 8);
    Wire.endTransmission();
    Wire.requestFrom(ALS_ADDRESS, 8);
    while (!Wire.available());
    Wire.readBytes(buff, 8);

    Serial.println(sensor0.readTempF());

    uint16_t pointX;
    uint16_t pointY;
    uint16_t type = 0;

    type = AXS_GET_GESTURE_TYPE(buff);
    pointX = AXS_GET_POINT_X(buff,0);
    pointY = AXS_GET_POINT_Y(buff,0);

    if (!type && (pointX || pointY)) {
        pointX = (640-pointX);
        if(pointX > 640) pointX = 640;
        if(pointY > 180) pointY = 180;

        printf("x = %d, y = %d\n", pointX, pointY);
    }
    delay(50);
}
```

lewisxhe, 2023-11-30:
It can be seen from [here](https://github.com/sparkfun/SparkFun_TMP102_Arduino_Library/blob/1260e0a9388918534504753323ae1df5a2ed1b3c/src/SparkFunTMP102.h#L34) that you did not pass Wire1 to TMP102 Class, so TMP102 uses Wire by default, you should use

```
uint8_t TMP102_address = 0x48;
sensor0.begin(TMP102_address , Wire1);
```

d3mac123, 2023-11-30:
I have tried your suggestion (thanks!) but I keep getting "Cannot connect to TMP102.". I guess there is something wrong with the pin setup or my board. I noticed that, when the board is connected to the USB cable, the TMP sensor LED turns on for a second then it turns off - the reason why is sending the "not connected" message. Any ideas on what is wrong here?

```
#include <Arduino.h>
#include <Wire.h>
#include "AXS15231B.h"

#include <SparkFunTMP102.h>
TMP102 sensor0;

uint8_t ALS_ADDRESS = 0x3B;
#define TOUCH_IICSCL 10
#define TOUCH_IICSDA 15
#define TOUCH_INT 11
#define TOUCH_RES 16

#define AXS_TOUCH_ONE_POINT_LEN             6
#define AXS_TOUCH_BUF_HEAD_LEN              2

#define AXS_TOUCH_GESTURE_POS               0
#define AXS_TOUCH_POINT_NUM                 1
#define AXS_TOUCH_EVENT_POS                 2
#define AXS_TOUCH_X_H_POS                   2
#define AXS_TOUCH_X_L_POS                   3
#define AXS_TOUCH_ID_POS                    4
#define AXS_TOUCH_Y_H_POS                   4
#define AXS_TOUCH_Y_L_POS                   5
#define AXS_TOUCH_WEIGHT_POS                6
#define AXS_TOUCH_AREA_POS                  7

#define AXS_GET_POINT_NUM(buf) buf[AXS_TOUCH_POINT_NUM]
#define AXS_GET_GESTURE_TYPE(buf)  buf[AXS_TOUCH_GESTURE_POS]
#define AXS_GET_POINT_X(buf,point_index) (((uint16_t)(buf[AXS_TOUCH_ONE_POINT_LEN*point_index+AXS_TOUCH_X_H_POS] & 0x0F) <<8) + (uint16_t)buf[AXS_TOUCH_ONE_POINT_LEN*point_index+AXS_TOUCH_X_L_POS])
#define AXS_GET_POINT_Y(buf,point_index) (((uint16_t)(buf[AXS_TOUCH_ONE_POINT_LEN*point_index+AXS_TOUCH_Y_H_POS] & 0x0F) <<8) + (uint16_t)buf[AXS_TOUCH_ONE_POINT_LEN*point_index+AXS_TOUCH_Y_L_POS])
#define AXS_GET_POINT_EVENT(buf,point_index) (buf[AXS_TOUCH_ONE_POINT_LEN*point_index+AXS_TOUCH_EVENT_POS] >> 6)

void setup() {
    Serial.begin(115200);
    Serial.println("sta\n");

    pinMode(TOUCH_RES, OUTPUT);
    digitalWrite(TOUCH_RES, HIGH);delay(2);
    digitalWrite(TOUCH_RES, LOW);delay(10);
    digitalWrite(TOUCH_RES, HIGH);delay(2);

    Wire.begin(TOUCH_IICSDA, TOUCH_IICSCL);
    Wire1.begin(43,44);
    uint8_t TMP102_address = 0x48;
    sensor0.begin(TMP102_address , Wire1);

    axs15231_init();  //If you use touch, you must initialize the screen first

    Serial.println("end\n");
}

uint8_t read_touchpad_cmd[11] = {0xb5, 0xab, 0xa5, 0x5a, 0x0, 0x0, 0x0, 0x8};
void loop() {
    uint8_t buff[20] = {0};
    Wire.beginTransmission(ALS_ADDRESS);
    Wire.write(read_touchpad_cmd, 8);
    Wire.endTransmission();
    Wire.requestFrom(ALS_ADDRESS, 8);
    while (!Wire.available());
    Wire.readBytes(buff, 8);

    Serial.println(sensor0.readTempF());

    uint16_t pointX;
    uint16_t pointY;
    uint16_t type = 0;

    type = AXS_GET_GESTURE_TYPE(buff);
    pointX = AXS_GET_POINT_X(buff,0);
    pointY = AXS_GET_POINT_Y(buff,0);

    if (!type && (pointX || pointY)) {
        pointX = (640-pointX);
        if(pointX > 640) pointX = 640;
        if(pointY > 180) pointY = 180;

        printf("x = %d, y = %d\n", pointX, pointY);
    }
    delay(50);
}
```

lewisxhe, 2023-12-01:
Is your line sequence correct? Can you take a photo and have a look?

d3mac123, 2023-12-01:
I found interesting things:

  *   I have tried 8 different qwiic cables (all of them work with the original T-Display) and none of them work. However, when I used the 2 small cables that came with the board, they worked;

  *   I then tried to use the 2 small cables (that came with the board and worked) with a Sparkfun Qwiic hub (https://www.sparkfun.com/products/18012) and the sensor did not work. However, removing the hub from the chain, the sensor returns to work;

  *   My next test was to connect another sensor to the TMP102 sensor chain (I did not code the sensor, just wanted to see if it gets power from the chain) and it did not work as well.

I guess my questions are:

  *   why things are working only with your small cables?
  *   why any chain configuration is not working?

SparkFun Qwiic MultiPort - BOB-18012 - SparkFun Electronics<https://www.sparkfun.com/products/18012>
The Qwiic MultiPort adds additional ports to boards that have only one Qwiic port on the I2C bus.
www.sparkfun.com

  *

Alex Souza

lewisxhe, 2023-12-01:
Then I understand the problem. The initial version of QWIIC had incorrect wiring sequence, so the accompanying cable must be used, which has been processed

d3mac123, 2023-12-01:
So, my board is not compatible with any other cables?

lewisxhe, 2023-12-02:
> So, the only way to chain multiple sensors is to connect the cable that came with my board with the Qwiic connector on the board, then use a regular cable from the sensor to another sensor? This was the only way I could connect 2 sensors in my tests.
> 
> Do I need to replace my board? I need to chain several sensors in my project and if my board has Qwiic issues, I need to get a new one from Lilygo.

Yes, you just need to use the included QWIIC cable at the outlet of the motherboard. If you need it, you can explain to customer service that QWIIC requires reverse wire sequence, and ask them to find me, and I will explain the situation to them.

closed 2023-12-30 by d3mac123