Xinyuan-LilyGO/T-Display-S3-Long #5: AXS15231B datasheet update

nikthefix, 2023-11-19:
Please upload an updated version of the AXS15231 datasheet if you have it.

The current datasheet shows that optional parameters in the DCTR register can be set to change scan configuration which would allow for a hardware screen rotate rather than the currently necessary soft rotate - but the documentation appears to be incomplete. 

Perhaps these are OTP config settings for factory use but I'd love to know for sure.

Thanks.

nikthefix, 2023-11-29:
Thanks Lewis. I suspected that it wasn't implemented for this display model. Never mind, soft rotate is working just fine.

I have a landscape mode TFT_eSPI demo for the Long which I'll upload in a couple of days. Just tidying up a bit.

nik

nikthefix, 2023-11-29:
Sorry Lewis, one last question:

The display init sequence in const static lcd_cmd_t axs15231b_qspi_init_new[]......

Is this redundant? Was it used for testing purposes? Does it configure the display in a different way?

Thanks again,

nik

lewisxhe, 2023-11-29:
That's unnecessary. The guy who maintained this warehouse before didn't delete this part.

nikthefix, 2023-11-29:
Great! Thanks.

lewisxhe, 2023-12-05:
The initialization part and color are the optimal solutions adjusted by the screen supplier. We cannot obtain the OTP part. It is confidential and I cannot help you.

bao55611, 2024-07-28:
hi nik ... can you make AXS15231B work with TFT ESPI by squareline studio ?

nikthefix, 2024-07-29:
@bao55611 

Yes I made a demo to do exactly this here:

https://github.com/nikthefix/Lilygo_Support_T_Display_S3_Long_LVGL_Odometer

I made the UI using SLS.

Dpaganini, 2024-10-05:
@nikthefix I'm able to made this work on a AXS15231B  at a jc3248w535 but not able to fill entire display (480 x 320). Do you have any tips?

nikthefix, 2024-10-05:
@Dpaganini 
My demo code was written for the Lilygo Long board which has a different pixel layout. To modify the sketch for your board you'll need to redefine the width / height parameters and make changes to the datums in the lvgl / sls code. It would probably be easier to start a fresh sls project and use my demo as a template into which you can import the new sls export files.

vencas, 2024-11-08:
@Dpaganini Can you share code with me, I have JC3248W535 from ali, tried define ports (but without luck)
```
#define TFT_QSPI_CS           45 //12
#define TFT_QSPI_SCK          47 //17
#define TFT_QSPI_D0           21 //13
#define TFT_QSPI_D1           48 //18
#define TFT_QSPI_D2           40 //21
#define TFT_QSPI_D3           39 //14
#define TFT_QSPI_RST          -1 //16 , I don't know this one
#define TFT_BL                1
```

Thank you

lukepasek, 2024-12-23:
I have JC3248W535C from Aliexpress and I managed to make it work quite nicely with Lvgl, and with fast hw fill, if anyone is interested I can share the code.

Jakany95, 2025-01-04:
> JC3248W535C
Hi, I have the same board, please if you can help me with the file it would be a great help.

bensuffolk, 2025-02-26:
I have a JC3248W535C which I have got to work with LVGL but I notice that the ESP LCD drivers have different init code to the demo that came with the board, which again is different to the code here. 

The data sheet explains what some of the commands are, but a lot of them are not listed in the sheet. How does one find out what the relevant init code commands are doing?

nikthefix, 2025-02-27:
@bensuffolk 
The display datasheet should list all the commands but there are a couple of points:

Often the publicly available datasheets are marked preliminary or v0.1 or similar and are sometimes incomplete.
Ask the display manufacturer or distributor for the latest documentation.

Also, in my experience, some of these cheaper dev boards use displays which are surplus stock originally intended for a specific product and have been one-time-programmed with a vendor configuration. The OTP pages are likely not shown in the datasheet as they are reserved for 'OEM eyes only'. It could be that the commands you see but don't recognize relate to this custom configuration.

Post any unknown command addresses here and I'll see if I can match them up with some of my collected datasheets.

nik

nikthefix, 2026-01-29:
@Dead4ever07 
I'll be able to take a look at this tomorrow.
But in the meantime, I forgot to ask, are you pushing to the display in native portrait mode or is your coordinate system targeting landscape? I suggest performing all experiments in portrait initially in order to avoid outside bounds errors when addressing the GRAM.

Dead4ever07, 2026-01-29:
@nikthefix 
I'm currently using my display in portrait mode. Let me know which conclusions did you arrived at.