Xinyuan-LilyGO/T-Display-S3-Long #8: Double tap happening all the time

d3mac123, 2023-12-08:
@nikthefix I am not sure if this is related to your library (I guess it is related to the `lcd_PushColors_rotated_90(0, 0, 640, 180, (uint16_t*)sprite.getPointer());` call but, I am not sure) but, every time I tap the display (for example, a tap would go to another screen), I am getting double returns. 

Explaining better, lets say the user is in screen1 and taps the display. He should go to screen2. However, what is happening is the code sends him to screen2 but returns to screen. Any ideas why?

nikthefix, 2023-12-09:
Hi,

How have you configured the 'tap to select screen' in your sketch?
Is it a button or just a tap anywhere on the panel?
Is it 'activate on press' or  'activate on release'?

I suggest starting with separate gui buttons for separate functions - just until everything seems to be working as it should.
Since the touch driver is always polling the sensor (at least in the Lilygo demo), touch functions at the same coordinates on 2 different screens may rebound causing the effect you describe. You end up with a kind of gui oscillator!

Activate on release can really help with this. 
Using the touch interrupt pin to limit polling to touch duration and detect a finger release is also a good thing.
Use software traps to compare and discard any unchanged polled coordinates.

I've noticed that some touch panels show coord 0,0 upon release. I guess this is to help in cases where there is no interrupt implementation but it can problematic if you don't set a trap for 0,0

I have touch working with LVGL and my modified graphic driver so I think the hardware and driver are OK so far. I'll try it with TFT_eSPI. Are you using TFT_eSPI touch implementation or rolling your own?

Cheers and thank you for the coffees.

nik

d3mac123, 2023-12-09:
I have to admit my development skills are not the best so, my answers will surprise you :)
- I just used the touch demo example - basically, I used lines 61-71 to address where the screen was touched, and then I redirected commands accordingly;
- I would love to control each touch (onRelease preferably) but, being my first time working with a touch screen, I couldn't find any examples on how to do that;
- I am not familiar on how to use/control the PINs.

Would you happen to have any basic examples on how to set up and capture a button touch with the TFT_eSPI?

Regarding the coffees, you deserve way more than that. Thanks again!

nikthefix, 2023-12-09:
Ah yes, the Lilygo example is a 'blunt instrument' implementation of touch. You have to do more to make it useful. I'll post an example asap but I think Volos will have something supreme in the next day or two which we can all learn from.

closed 2023-12-30 by d3mac123