Xinyuan-LilyGO/T-Display-S3-Long #9: How to use both Qwiic ports?

d3mac123, 2023-12-11:
Hi, 

How to use both I2C ports (plus the touch part)? I can connect to one port using the Wire1 option (see below), but I am not sure on how to also enable the second port (pins 18 and 15?).

```
void setup() {
    Serial.begin(115200);
    Serial.println("sta\n");

    pinMode(TOUCH_RES, OUTPUT);
    digitalWrite(TOUCH_RES, HIGH);delay(2);
    digitalWrite(TOUCH_RES, LOW);delay(10);
    digitalWrite(TOUCH_RES, HIGH);delay(2);

    Wire.begin(TOUCH_IICSDA, TOUCH_IICSCL);
    Wire1.begin(43,44);
    uint8_t TMP102_address = 0x48;
    sensor0.begin(TMP102_address , Wire1);

    axs15231_init();  //If you use touch, you must initialize the screen first

    Serial.println("end\n");
}
```

lewisxhe, 2023-12-12:
For the touch group of I2C, you only need to initialize the touch, and Qwiic can go up directly.

d3mac123, 2023-12-12:
Not sure I understood your reply: how to enable my second port? My sensors need to have the Wire command enabling the proper pins - for the first Qwiic connector, I use Wire1.begin(43,44));

lewisxhe, 2023-12-13:
I added an example, hope it helps you

closed 2023-12-30 by d3mac123