Xinyuan-LilyGO/T-Display-S3-Long #24: SY6970 library does not change the actual charging rate.

Grex44, 2024-04-02:
Hi, 

I'm experiencing some issue with the charging speed rate on the SY6970. I included the library from lilygo's repository. Even after setting up  _PMU.setChargerConstantCurr();_ to _PMU.setChargerConstantCurr(3200);_ The tested charging rate was still at 400mA. 

I've even increase the input current limit with _setInputCurrentLimit()_ to _setInputCurrentLimit(2000)_; 

There was no change in the actual charging performance.
 Appreciate your assistance in it.

rhcp4life, 2024-04-02:
is the charging rate fixed based on this resistor?
[image]

Grex44, 2024-04-03:
Ok thanks. Is there anyone possible to point out which resistor is R24 on the board itself since there isn't any silkscreen indicating R24 

[image]

lewisxhe, 2024-04-03:
Do you want to set the charging current to 3A? I do not recommend adjusting the charging current. It is reasonable to set the maximum to 500mA according to the circuit conditions.

Grex44, 2024-04-03:
Yes or anything more than 500mA. I would like to increase the charging speed time of the battery that is currently being used. Maybe increase it to 1A or 2A at most.

lewisxhe, 2024-04-03:
Please note that replacing this resistor carries the risk of damage, which you should be aware of
The picture below shows the location of the resistor
[image]

Grex44, 2024-04-05:
Thanks for the assistance. I will follow up again if there is any results upon the swap of resistor.