Xinyuan-LilyGO/T-Display-S3-Long #31: Disable Green LED when using ESP Home.

pixelwave, 2024-06-18:
Is there a way to disable the blinking green LED? 

I am using ESP Home ... maybe by pulling one GPIO pin low or something?

```
substitutions:
  comment: "Use to change desk height and switch active PC in basement."
  device_name: esp32-07-desk-control
  friendly_name: "(BASE) Desk Control Display"

esphome:
  name: ${device_name}
  friendly_name: ${friendly_name}
  comment: ${comment}
  platformio_options:
    upload_speed: 921600
    build_unflags: -Werror=all
    board_build.flash_mode: dio
    board_build.f_flash: 80000000L
    board_build.f_cpu: 240000000L

psram:
  mode: octal
  speed: 120MHz

esp32:
  board: esp32-s3-devkitc-1
  flash_size: 16MB
  framework:
    type: esp-idf

external_components:
  - source: github://buglloc/esphome-components
    components: [axs15231]

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "..."
  services:
    - service: turn_off
      then:
        - script.execute: turn_display_off

ota:
  password: "..."

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "T-Display-S3-Long"
    password: "..."

captive_portal:

spi:
  - id: display_qspi
    type: quad
    clk_pin: 17
    data_pins:
      - 13
      - 18
      - 21
      - 14

i2c:
  sda: 15
  scl: 10
  id: touchscreen_bus

font:
  - file: "gfonts://Roboto"
    id: font_std
    size: 40
    glyphs: "!\"%()+=,-_.:°0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnopqrstuvwxyz/\\[]|&@#'"
  - file: "gfonts://Roboto@700"
    id: font_title
    size: 40
    glyphs: "!\"%()+=,-_.:°0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnopqrstuvwxyz/\\[]|&@#'"

globals:
  - id: bgcolor
    type: Color
    #initial_value: "Color::random_color()"

script:
  # Turn the display off
  - id: turn_display_off
    then:
      - light.turn_off: back_light
      - display.page.show: page_off
      - logger.log: "Display turned OFF"

color: 
  - id: red
    hex: FF0000
  - id: blue
    hex: 0000FF
  - id: black
    hex: '000000'

image:
  - file: "desk_control.png"
    id: img_desk_control
    type: RGBA
    resize: 640x180
  - file: "pc_control.png"
    id: img_pc_control
    type: RGBA
    resize: 640x180

display:
  - platform: axs15231
    id: main_display
    spi_id: display_qspi
    dimensions:
      height: 640
      width: 180
    cs_pin: 12
    reset_pin: 16
    #backlight_pin: 1
    backlight_pin: 
      number: GPIO1
      allow_other_uses: true  
    transform:
      swap_xy: false
    rotation: 270
    auto_clear_enabled: false
    pages:
      - id: page_desk_control
        lambda: |-
          // it.print(it.get_width()/2, it.get_height()/2-20, id(font_title), TextAlign::CENTER, "Desk Control");
          // it.print(it.get_width()/2, it.get_height()/2+20, id(font_std), TextAlign::CENTER, "PAGE 1");
          // Draw the image my_image at position [x=0,y=0]
          it.image(0, 0, id(img_desk_control));
      - id: page_pc_control
        lambda: |-
          it.image(0, 0, id(img_pc_control));
      - id: page_off
        lambda: |-
          it.fill(id(black));

touchscreen:
  - platform: axs15231
    id: main_touch
    display: main_display
    i2c_id: touchscreen_bus
    transform:
      swap_xy: true
    on_touch:
      - lambda: |-
          ESP_LOGI("cal", "x=%d, y=%d, x_raw=%d, y_raw=%0d",
            touch.x,
            touch.y,
            touch.x_raw,
            touch.y_raw
          );   

binary_sensor:
  # Switch Button
  - platform: touchscreen
    name: Page_Switch_Desk
    internal: true
    x_min: 0
    x_max: 180
    y_min: 540
    y_max: 640
    page_id: page_desk_control
    on_multi_click:
    - timing:
        - ON for at least 2s
        #- OFF for at least 0.5s
      then:
        - logger.log: "Multi Button -> Single Long Clicked"
        - script.execute: turn_display_off
    - timing:
        - ON for at most 1s
        - OFF for at least 0.5s
      then:
        - logger.log: "Multi Button -> Single Short Clicked"
        #- display.page.show_next: main_display
        - display.page.show: page_pc_control
  - platform: touchscreen
    name: Page_Switch_PC
    internal: true
    x_min: 0
    x_max: 180
    y_min: 540
    y_max: 640
    page_id: page_pc_control
    on_multi_click:
    - timing:
        - ON for at least 2s
        #- OFF for at least 0.5s
      then:
        - logger.log: "Multi Button -> Single Long Clicked"
        - script.execute: turn_display_off
    - timing:
        - ON for at most 1s
        - OFF for at least 0.5s
      then:
        - logger.log: "Multi Button -> Single Short Clicked"
        #- display.page.show_next: main_display
        - display.page.show: page_desk_control

  # Buttons for: page_desk_control
  - platform: touchscreen
    name: DKC_BTN_1
    x_min: 0
    x_max: 180
    y_min: 0
    y_max: 180
    page_id: page_desk_control
    on_press:
      - logger.log: "DKC_BTN_1 was touched in Desk Control."
      #- display.page.show: page_pc_control
  - platform: touchscreen
    name: DKC_BTN_2
    x_min: 0
    x_max: 180
    y_min: 180
    y_max: 360
    page_id: page_desk_control
    on_press:
      - logger.log: "DKC_BTN_2 was touched in Desk Control."
  - platform: touchscreen
    name: DKC_BTN_3
    x_min: 0
    x_max: 180
    y_min: 360
    y_max: 540
    page_id: page_desk_control
    on_press:
      - logger.log: "DKC_BTN_3 was touched in Desk Control."

  # Buttons for: page_pc_control
  - platform: touchscreen
    name: PCC_BTN_1
    x_min: 0
    x_max: 180
    y_min: 0
    y_max: 180
    page_id: page_pc_control
    on_press:
      - logger.log: "PCC_BTN_1 was touched in Desk Control."
      #- display.page.show: page_pc_control
  - platform: touchscreen
    name: PCC_BTN_2
    x_min: 0
    x_max: 180
    y_min: 180
    y_max: 360
    page_id: page_pc_control
    on_press:
      - logger.log: "PCC_BTN_2 was touched in Desk Control."
  - platform: touchscreen
    name: PCC_BTN_3
    x_min: 0
    x_max: 180
    y_min: 360
    y_max: 540
    page_id: page_pc_control
    on_press:
      - logger.log: "PCC_BTN_3 was touched in Desk Control."

  # Buttons for: page_off
  - platform: touchscreen
    name: display_on
    internal: true
    x_min: 0
    x_max: 180
    y_min: 0
    y_max: 540
    page_id: page_off
    on_press:
      - logger.log: "BTN_DISPLAY_ON was touched."
      - light.turn_on: back_light
      - display.page.show: page_desk_control

# Define Backlight as Switch to turn display on/off
output:
  #Backlight pwm section
  - platform: ledc
    pin:
      number: GPIO1
      allow_other_uses: true  
    id: backlight_pwm
    #inverted: true
       
light:
  - platform: monochromatic
    output: backlight_pwm
    name: "Backlight"
    id: back_light
    restore_mode: ALWAYS_ON
    internal: false
```

fundix, 2024-06-18:
`#include "PMU.h"

    PowersSY6970 PMU_X;

    result = PMU_X.init(Wire, TOUCH_IICSDA, TOUCH_IICSCL, SY6970_SLAVE_ADDRESS);
    if (result == false)
    {
        Serial.println("PMU is not online...");
        delay(50);
    }
    else
    {
        PMU_X.enableADCMeasure();
        PMU_X.disableStatLed();
        PMU_X.disableOTG();
    }
    PMU_X.deinit();`

pixelwave, 2024-06-19:
> ```
> #include "PMU.h"
>
>     PowersSY6970 PMU_X;
>
>     result = PMU_X.init(Wire, TOUCH_IICSDA, TOUCH_IICSCL, SY6970_SLAVE_ADDRESS);
>     if (result == false)
>     {
>         Serial.println("PMU is not [...]

How would that work in ESPHome?

fundix, 2024-06-20:
> How would that work in ESPHome?

no code is from Platformio - arduino framework

pixelwave, 2024-06-20:
What I thought ... I am still learning. Not sure how to translate / integrate this into ESP Home code.

roderik-maker, 2024-08-19:
Would also love to find a solution to this.

buglloc, 2024-11-07:
If you are still searching for a way to disable the green state LED in ESPHome, I have pushed a simple component for this purpose: https://github.com/buglloc/esphome-components?tab=readme-ov-file#sy6970-pmu-wip