Xinyuan-LilyGO/T-Display-S3-Long #34: Random touch read

fundix, 2024-07-15:
I need to solve the random clicking on the display. I have several boards and they all do it at different intervals. the click area is always just from this area
```
const uint16_t MIN_X = 530;
const uint16_t MAX_X = 600;
const uint16_t MIN_Y = 100;
const uint16_t MAX_Y = 150;
```

fundix, 2024-07-16:
@LilyGO do you have any solution for this problem

Llgok, 2024-07-20:
@fundix 
Hello, may I ask if the screen randomly clicks automatically when the finger is not touching it?

fundix, 2024-07-20:
> @fundix Hello, may I ask if the screen randomly clicks automatically when the finger is not touching it?

Yes, we have several Long units, the one I have on my desk now gerates random clicks in this area relatively frequently for around 2minutes. We are using the LVGL library from your example. I'll take a look at issue #30 you advise there

Llgok, 2024-07-20:
@fundix 
You can try using the [GFX\_AXS15231B\_Image](https://github.com/Xinyuan-LilyGO/T-Display-S3-Long/tree/master/examples/GFX_AXS15231B_Image) program to see if it mistakenly triggers without any finger touch.

malexnurazhar, 2024-08-15:
Im having the same issue. Random touch in the same area without touching.

fundix, 2024-08-15:
@malexnurazhar 
> Im having the same issue. Random touch in the same area without touching.

Do you use WiFi in your project? Try disabling it and test if the random clicking remains. In my project, it seems that when I removed the code that manages the wifi, the clicking either disappeared or hasn't appeared yet. But with wifi it was quite partial.

malexnurazhar, 2024-08-15:
No wifi no bluetooth. Just the display and touch. i tried using interrupts as suggested to only read i2c when theres touch. Didnt really affect anything. I suspect its something to do with the power. Its very sensitive. The issue got worse with a noisy power supply. Its better when powered by battery. But problem still persist every few seconds.

malexnurazhar, 2024-08-15:
> @malexnurazhar
> 
> > Im having the same issue. Random touch in the same area without touching.
> 
> Do you use WiFi in your project? Try disabling it and test if the random clicking remains. In my project, it seems that when I removed the code that manages the wifi, the clicking either disappeared or hasn't appeared yet. But with wifi it was quite partial.
> 
> update: Update: No, so the problem will be as someone else wrote with what is displayed on the screen. The all white area causes me to click even without wifi.
> 
> [107915][I][Display.cpp:297] my_touchpad_read(): [Display] fingers_number: 2, touch event 8 [108065][I][Display.cpp:297] my_touchpad_read(): [Display] fingers_number: 2, touch event 8 [108215][I][Display.cpp:297] my_touchpad_read(): [Display] fingers_number: 2, touch event 8 [108365][I][Display.cpp:297] my_touchpad_read(): [Display] fingers_number: 2, touch event 8 ...

I suspect something to do with the power supply. Its a very power sensitive display. Not sure if my theory makes sense. When its white, a small voltage is applied to the pixel to turn all the film to transparent. This small voltage might have leak to the capacitive panel causing it to detect touch input.

malexnurazhar, 2024-08-15:
But mine is clicking randomly even with all black.

Llgok, 2024-08-15:
> But mine is clicking randomly even with all black.

@malexnurazhar 
Could you provide us with your complete project testing code? This would be more helpful in assisting us to identify any issues.

fundix, 2024-08-15:
I'll try to get the project to compile without our other libraries

display ok
[image]

display not ok
[image]

[196146][I][Display.cpp:297] my_touchpad_read(): [Display] fingers_number: 2, touch event 8
[196249][I][Display.cpp:297] my_touchpad_read(): [Display] fingers_number: 2, touch event 8
[196353][I][Display.cpp:297] my_touchpad_read(): [Display] fingers_number: 2, touch event 8
[196505][I][Display.cpp:297] my_touchpad_read(): [Display] fingers_number: 2, touch event 8

fundix, 2024-08-16:
> > But mine is clicking randomly even with all black.
> 
> @malexnurazhar Could you provide us with your complete project testing code? This would be more helpful in assisting us to identify any issues.

@Llgok I added you to our repository branch Touch-test
https://github.com/Liftor-GmbH/Liftor-Ortex-Remote/tree/Touch-test

the display goes off after about 30sec sometimes it lights up again

[872553][I][Display.cpp:190] display_wakeup_event_handler(): [Display] wakeup_event_handler - screen pressed
[872570][I][Display.cpp:296] my_touchpad_read(): [Display] fingers_number: 1, touch event 4
[906839][I][Display.cpp:296] my_touchpad_read(): [Display] fingers_number: 1, touch event 8
[906852][I][Display.cpp:296] my_touchpad_read(): [Display] fingers_number: 0, touch event 0
[906861][I][Display.cpp:190] display_wakeup_event_handler(): [Display] wakeup_event_handler - screen pressed

Llgok, 2024-08-19:
@fundix 
I tested the code you provided and recorded a video for you to take a look at. It seems that there was no accidental screen wake-up trigger during the 30-second screen-off test.

https://github.com/user-attachments/assets/809dd550-169e-4253-8fd2-a9b791a67b71

Llgok, 2024-08-19:
I will move the test output information function down to the triggering range.
[image]

malexnurazhar, 2024-08-19:
All 3 of my unit is showing the same issue even with filters. So i disabled the touch input at the problematic area for the time being.
If(pointX > 500 && pointY > 100)return;
Will wait for the new ones to come. To see if its a faulty unit.

fundix, 2024-08-19:
> @fundix I tested the code you provided and recorded a video for you to take a look at. It seems that there was no accidental screen wake-up trigger during the 30-second screen-off test.
> 
>  touch.test.mp4

@Llgok 

Thank you for the test. The problem for me arises later The display turns itself off after 30sec (the time can be changed in the swipe up settings) and lights up at different time intervals.  I shortened the video, it took around 1min till the light up.

[image]

https://github.com/user-attachments/assets/8b63a1b2-9621-4260-8207-32054bb687ef

malexnurazhar, 2024-08-19:
> All 3 of my unit is showing the same issue even with filters. So i disabled the touch input at the problematic area for the time being. If(pointX > 500 && pointY > 100)return; Will wait for the new ones to come. To see if its a faulty unit.

I got the new unit and the random touch isnt there on the new unit. Does it mean the one with the random touch has hardware issue??. Now the new unit has a new issue, i need to manually press boot and reset every single time to upload my code.

Llgok, 2024-08-20:
@fundix 
You can try adding the following lines for power initialization parameters in your code:

Touch-test branch
liftor-remote.cpp[39,0]
```C++
bool WriteC8D8(uint8_t c, uint8_t d)
{
    if (Wire.write(c) == false)
    {
        log_e("->Write(c) fail");
        return false;
    }
    if (Wire.write(d) == false)
    {
        log_e("->Write(d) fail");
        return false;
    }
    return true;
}

bool IIC_WriteC8D8(uint8_t device_address, uint8_t c, uint8_t d)
{
    Wire.beginTransmission(device_address);
    if (WriteC8D8(c, d) == false)
    {
        log_e("->WriteC8D8(c, d) fail");
        return false;
    }
    if (!(Wire.endTransmission()) == false)
    {
        log_e("->EndTransmission() fail");
        return false;
    }
    return true;
}
```

Touch-test branch
liftor-remote.cpp[87,0]
```C++
    //Disable the ILIM pin and set the input current limit to maximum.
    IIC_WriteC8D8(0x6A, 0x00, 0B00111111);
    //Turn off the BATFET without using the battery.
    IIC_WriteC8D8(0x6A, 0x09, 0B01100100); 
```
[image]
[image]

Check if the random touch issue disappears after adding the above code.

fundix, 2024-08-22:
@Llgok 
Looks like that solves the problem. Thank you very much.