Xinyuan-LilyGO/T-Display-S3-Long #36: Unable to flash without manually pressing reset and boot

malexnurazhar, 2024-08-19:
I need to manually press boot and reset everytime i upload. This was not required on my previous s3 long. Tried cdc on boot. Nothing works

Llgok, 2024-08-20:
@malexnurazhar 
May I ask what version of arduino esp lib you are using?

malexnurazhar, 2024-08-20:
> @malexnurazhar May I ask what version of arduino esp lib you are using?

PACKAGES:
 - framework-arduinoespressif32 @ 3.20014.231204 (2.0.14)
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5

Llgok, 2024-08-20:
@malexnurazhar 
Have you turned off USB mode? If USB mode is enabled, you won't be able to upload programs via the USB port. You need to enter boot mode to download.
[image]

malexnurazhar, 2024-08-20:
> @malexnurazhar Have you turned off USB mode? If USB mode is enabled, you won't be able to upload programs via the USB port. You need to enter boot mode to download. ![1](https://private-user-images.githubusercontent.com/135582120/359438056-37eb61e1-d3db-4b0d-83fc-7f56e8da5b67.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQxNTg2NzUsIm5iZiI6MTcyNDE1ODM3NSwicGF0aCI6Ii8xMzU1ODIxMjAvMzU5NDM4MDU2LTM3ZWI2MWUxLWQzZGItNGIwZC04M2ZjLTdmNTZlOGRhNWI2Ny5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwODIwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDgyMFQxMjUyNTVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03NmVkNWNjYzAyZGRhMTQ2N2NiNDhjMWQwMzc0OTEwYjM5OTVkMTZjODMyMWMxY2MxZDU4MjRmYWM5OTYyM2QyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.lpt8D5QzMSRtP0pR-hPROMucUQVqiqu-S75RoIGkkTk)

turned off, still unable to upload normally. is there a hardware change between the old display and the new display? alll my previous unit have no issue uploading normally

Llgok, 2024-08-21:
@malexnurazhar 
You can try upgrading the arduino esp lib version to 2.0.17 before proceeding.
[image]

malexnurazhar, 2024-08-21:
> @malexnurazhar You can try upgrading the arduino esp lib version to 2.0.17 before proceeding. ![1](https://private-user-images.githubusercontent.com/135582120/359715059-c96f467c-6450-47f9-926d-0f71102c03aa.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQyMTEyNzMsIm5iZiI6MTcyNDIxMDk3MywicGF0aCI6Ii8xMzU1ODIxMjAvMzU5NzE1MDU5LWM5NmY0NjdjLTY0NTAtNDdmOS05MjZkLTBmNzExMDJjMDNhYS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwODIxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDgyMVQwMzI5MzNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02YWQwMWUyMjg0NDdhMzA2NjRlM2FhMWJlMWQwMzk0YTM3YTE3ZTBmNTJhNDNlNmZmZWMyNDQ3YzdlYjcyZDcxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.jt7ZNpoDz1rXTG3YPHDX_xVxkkcMvGvq0h6vpWOpvXE)

In arduino ide, 2.0.17 doesnt work. Version 3.0.0 to 3.0.4 working perfectly. In platformio highest version is 2.0.17. Development version doesnt work.

Llgok, 2024-08-21:
@malexnurazhar 
May I ask which testing program you are using?

Llgok, 2024-08-21:
@malexnurazhar 
If you are using arduino esp lib version 3.0.0 or above, you may encounter a bug where you need to hold down the boot button during upload. This is an issue with the library that requires an update from the Arduino community. You can ask for help in their GitHub Q&A section, [arduino-esp32 issues](https://github.com/espressif/arduino-esp32/issues).
﻿
Version 2.0.14 will not have this problem. In version 2.0.14, you can try using [GFX\_AXS15231B\_Image](https://github.com/Xinyuan-LilyGO/T-Display-S3-Long/tree/master/examples/GFX_AXS15231B_Image) to test it. Other examples may fail to compile if updated to version 3.0.0 or above when compiled with version 2.0.0.

iotricity, 2024-09-27:
This is a known issue for the ESP32 S3 with direct USB. On some USB ports/host chips, the automatic reset doesn't function properly. Try another USB hub or port.