Xinyuan-LilyGO/T-Display-S3-Long #38: Arduino code ESP.reset() will reset board but display does not come back on.

tavdog, 2024-09-12:
ESP.reset() will reset board but display does come back on.  Manual press of hardware reset button is required to get the display to come back on during reboot.

Is there some sort of display reset function needed to be called somewhere ?

fundix, 2024-09-12:
reset is e.g. for the init function of the display driver
but in the driver documentation there are also functions for sleep and reset

```
void axs15231_init(void)
{
    pinMode(TFT_QSPI_CS, OUTPUT);
    pinMode(TFT_QSPI_RST, OUTPUT);

    TFT_RES_H;
    delay(130);
    TFT_RES_L;
    delay(130);
    TFT_RES_H;
    delay(300);
    ...
    
 void lcd_sleep()
{
    lcd_send_cmd(0x10, NULL, 0);
}
```

tavdog, 2024-09-13:
I am calling axs15231_init(); during boot but it doesn't work after calling ESP.restart().

tavdog, 2024-09-13:
I managed to get a software restart to work by doing the deep sleep for 0 method :
```
esp_sleep_enable_timer_wakeup(); // Set wake-up time to 10 seconds
esp_deep_sleep_start(); 
```

closed 2024-09-16 by tavdog