Xinyuan-LilyGO/T-Display-S3-Long #39: Orange glow (backlight bleed?) around the screen

Hannah-GBS, 2024-10-02:
I just bought a Long, and as soon as it arrived I flashed EspHome on it, and used buglloc's [display component](https://github.com/buglloc/esphome-components) with it, which works great.

However (and I noticed this in the initial boot up *before* I flashed it), it has this horrible orange glow around the edge of the screen. It's only visible if the screen is set to a dark colour (my use case is black, so that's a big issue). It seems like the backlight is bleeding through. Is my unit defective?

[image]

lewisxhe, 2024-10-04:
This doesn't look like a backlight, but a display effect. You can test it by flashing LilyGo's [factory firmware](https://github.com/Xinyuan-LilyGO/T-Display-S3-Long/tree/master/firmware)

Hannah-GBS, 2024-10-04:
I have flashed a couple different firmwares and the effect remains. Those border pixels do recolour when told to, and as mentioned the glow is most visible when they're set to dark colours.

lewisxhe, 2024-10-04:
Flash the factory firmware, send it to the dealer, explain the situation, if it is a screen problem, they will replace it with a new one for you

Hannah-GBS, 2024-10-04:
Just posting for posterity with the factory firmware, I will be contacting the seller (Lilygo Aliexpress store) when the National Holiday week is finished in China.
[image]

closed 2024-10-04 by Hannah-GBS