Xinyuan-LilyGO/T-Display-S3-Long #41: Powe Supply

lgomesfsp, 2024-11-13:
Can I connect a 5V power supply directly to the battery connector?

fundix, 2024-11-13:
We tried this as a power supply alternative and it worked. But in the end we chose a super capacitor instead of a battery. They set only 200mA on the charging winder and thus solved the problems with rests when connecting or scanning wifi.

lewisxhe, 2024-11-14:
5V power supply Please connect to the 5V Pin, do not connect to the battery connector, otherwise it will damage the power chip

lgomesfsp, 2024-11-14:
I'm making a simple cluster for my car, can I keep this 5V supplied by a LM7805 permanent on the pin to power the board?

lewisxhe, 2024-11-15:
Of course you can, but 7805 is a linear regulator, so be sure to add a heat sink or use a DC step-down to reduce temperature rise.