Xinyuan-LilyGO/T-Display-S3-Long #42: Power consumption (deep sleep) and recovering from drained battery

phooji, 2024-12-21:
I was hoping to understand more about power management on this board. I'm running the board off of a single 18650 cell and hoping to maximize battery life. I'm using the  'wake on motion' interrupt from an accelerometer to wake from deep sleep, so ideally I'd like to power the I2C bus, the ESP32 itself, and nothing else.

* What's the best way to minimize power consumption while in deep sleep? At the moment I'm turning off WiFi/Bluetooth, setting the backlight pin to LOW, and putting the display to sleep, then putting the ESP32 into deep sleep. Is there more I could/should be doing?

* If the battery drains completely while the device is in deep sleep, it appears to put the PMU in a state where it refuses to charge the battery, and also doesn't boot the device, once plugged in again. Is there a setting I can change to ensure the PMU prevents or recovers from accidental battery drain?

I recognize these are more questions rather than real bugs,  but it'd be great to see some examples / documentation to cover this! Thanks!

phooji, 2025-01-02:
Happy new year! Still pretty interested in this!

lewisxhe, 2025-01-06:
1. The best way to sleep is to turn off all peripherals that can be turned off. You have mentioned it, and I think you did a good job.

2. Refusal to charge? Did you turn on the PMU to charge? Because it can be turned off, even a completely over-discharged battery can be fully activated and charged by the PMU.