Xinyuan-LilyGO/T-Display-S3-Long #44: Display distortion in slightly modified TFT sample code

Siroyan, 2025-02-11:
(Japanese is below.)

Hello,

I am using the sample source code provided in this repository ([here](https://gist.github.com/Siroyan/72804db2c2011ae479a82c9eb7fdad70)) for the T-Display-S3-Long module, and I have encountered the following display issues:

Display distortion:
- In some areas where a green line is expected, the line is missing.
- Certain regions that should be filled in red appear incomplete.
- Other than the attached script, no modifications have been made.
- Please refer to the attached photo for details.

I would appreciate it if you could take a look and advise on a possible solution. Thank you very much for your attention and support.

[image]
----------------------

公開されているサンプルソースコード（[こちら](https://gist.github.com/Siroyan/72804db2c2011ae479a82c9eb7fdad70)）を使用したところ、液晶表示に以下の不具合が発生しました。

表示の乱れについて
- 緑色の直線が描画されるはずの部分で線が引かれていない。
- 赤色に塗りつぶされるべき領域に欠けがある。
- 添付のスクリプト以外は全く変更していません。

lewisxhe, 2025-02-12:
Have you tested the code in this repository? Is it normal? If the same problem occurs when testing the code in this repository, the screen is probably damaged.

lewisxhe, 2025-02-13:
If this is the case, I suspect it has something to do with the start and end coordinates. Some amoleds cannot set odd x and y coordinates. Try changing the write coordinates and keep the coordinates even.

Siroyan, 2025-02-16:
Thank you for your helpful advice.
I tried modifying the code as follows: I changed the width and height arguments in `lcd_PushColors` to odd numbers, thinking that this would cause `lcd_address_set` to use even coordinates.
However, the issue with parts of the red area missing does not seem to be fixed.

``` hoge.c
// Fill red
lcd_PushColors(0,  0,  10, 360, (uint16_t*)fill_mem_all);    // Original
lcd_PushColors(0,  0,   9, 359, (uint16_t*)fill_mem_all);
```
``` hoge.c
// Green line
lcd_PushColors(5, 10,   1, 128, (uint16_t*)data1);    // Original
lcd_PushColors(5, 10,   1, 128, (uint16_t*)data1);
```
[image]

lewisxhe, 2025-02-17:
Try to cancel the definition of LCD_SPI_DMA in line 7 of AXS15231B.h

Siroyan, 2025-02-17:
Thank you for your reply.

I'm sorry for asking so many questions repeatedly, but I really like this device with its horizontally elongated display (which is such a rare feature!) and I absolutely want to get it working properly.

Following your advice, I turned off the DMA feature, but the issue did not improve.

I forked this repository and made modifications to clearly highlight the differences between my changes and the master branch. If possible, could you test the same source code on your device to see if the same issue occurs? This will help determine whether the problem is due to a hardware failure or a software bug.

[https://github.com/Siroyan/T-Display-S3-Long/commit/797430cbe4c5c88c9d83af4b59e63c7ab7b90327](url)

fundix, 2025-02-17:
display setup

```
    pinMode(TOUCH_INT, INPUT_PULLUP);

    ESP_LOGV(TAG, "Starting display br: %d timeout: %d", displayBrightness, displaySleepTime);

    axs15231_init();
    lv_init();
    vTaskDelay(pdMS_TO_TICKS(10));

    size_t buffer_size =
        sizeof(lv_color_t) * EXAMPLE_LCD_H_RES * EXAMPLE_LCD_V_RES;
    buf = (lv_color_t *)ps_malloc(buffer_size);
    assert(buf);

    buf1 = (lv_color_t *)ps_malloc(buffer_size);
    assert(buf1);

    lv_disp_draw_buf_init(&draw_buf, buf, buf1, buffer_size);
    /*Initialize the display*/
    static lv_disp_drv_t disp_drv;
    lv_disp_drv_init(&disp_drv);
    /*Change the following line to your display resolution*/
    disp_drv.hor_res = EXAMPLE_LCD_H_RES;
    disp_drv.ver_res = EXAMPLE_LCD_V_RES;
    disp_drv.flush_cb = my_disp_flush;
    disp_drv.draw_buf = &draw_buf;
    disp_drv.sw_rotate = 1; // If you turn on software rotation, Do not update or replace LVGL
    disp_drv.rotated = LV_DISP_ROT_90;
    disp_drv.full_refresh = 1; // full_refresh must be 1
    lv_disp_drv_register(&disp_drv);

    static lv_indev_drv_t indev_drv;
    lv_indev_drv_init(&indev_drv);
    indev_drv.type = LV_INDEV_TYPE_POINTER;
    indev_drv.read_cb = my_touchpad_read;
    indev_drv.scroll_limit = 15; // deftault  10

    lv_indev_drv_register(&indev_drv);

    ui_init();

    ledcAttach(TFT_BL, LEDC_BASE_FREQ, LEDC_TIMER_BIT);
    ledcWrite(TFT_BL, 1);
```

display flush
```
void my_disp_flush(lv_disp_drv_t *disp, const lv_area_t *area,
                   lv_color_t *color_p)
{
    uint32_t w = (area->x2 - area->x1 + 1);
    uint32_t h = (area->y2 - area->y1 + 1);

#ifdef LCD_SPI_DMA
    char i = 0;
    while (get_lcd_spi_dma_write())
    {
        i = i >> 1;
        lcd_PushColors(0, 0, 0, 0, NULL);
    }
#endif
    lcd_PushColors(area->x1, area->y1, w, h, (uint16_t *)&color_p->full);

#ifdef LCD_SPI_DMA

#else
    lv_disp_flush_ready(disp);
#endif
[... 571 lines cut ...]
        t.base.length = chunk_size * 16;

        spi_device_polling_transmit(spi, (spi_transaction_t *)&t);
        len -= chunk_size;
        p += chunk_size;
    } while (len > 0);
    TFT_CS_H;

#else
    TFT_CS_L;
    SPI.beginTransaction(SPISettings(SPI_FREQUENCY, MSBFIRST, TFT_SPI_MODE));

    SPI.writeBytes((uint8_t *)data, len * 2);
    SPI.endTransaction();
    TFT_CS_H;
#endif
}

void lcd_sleep()
{
    lcd_send_cmd(0x10, NULL, 0);
}
```

[image]